pachi81/GlucoDataHandler #216: Issue with BG Trend Arrow When Using Exponential Smoothing on AAPS

AhmedSalehz, 2025-05-08:
Hello,

Thank you for providing such an efficient application, which I’m currently using for alarms and notifications.

I have set AndriosAPS as the local source and configured AAPSClient to share data using Samsung Tizen. The data is being transmitted correctly. However, when the main smoothing method in AAPS is set to Exponential, the GlucoDataHandler consistently displays a question mark instead of the BG trend arrow.

Am I missing a configuration step or is this a known limitation?

Best regards,
Ahmed Saleh

pachi81, 2025-05-08:
No, this is an issue in AAPS.
For any reason it does not provide trend information in some cases...
As GDH has no historical data, it can not calculate a trend.
But in the next version, GDH can calculate the trend for such problems.
Maybe you can open an issue for AAPS...

reijoahola-nightscout, 2025-05-08:
I have now GDH v. 2.0 (Beta) in use and in AAPS turned Exponential filtering on. I do not see now ?-mark in Trend arrow. I remember that it was always on in these situations, Also Internal Trend calculation is turned off.

Has something changed in GDH v. 2.0 so that the problrem is fixed ?

AhmedSalehz, 2025-05-08:
how do i get beta version?

AhmedSalehz, 2025-05-08:
> As there is no bug in GDH, there is nothing, I can/could fix... The problem/bug is, that AAPS does include the trend value for special config. Just open a bug in AAPS. But in GDH v2 I added the trend calculation, which will be used if enabled or the source does not provide this information.

Yes, i already opened the issue at AAPS. I want to test the new trend calculation in GDH v2. How do i get this beta version?

reijoahola-nightscout, 2025-05-08:
I could not repeat the problem with GDH v. 2.0. In AAPS I had Exponential smoothing. In GDH internal Trend calculation was turned off. I know this ?-mark problem when using earlier GDH versio but something has fixed in GDH v. 2.0.

pachi81, 2025-05-08:
>  Trend calculation was turned off.

Even if you turn it off, it will used, as long the source does not provide a trend. So this setting is ignored, if AAPS does not provide a trend.

pachi81, 2025-05-08:
> How do i get this beta version?

Beta testing is an open testing, users can assign themself in playstore.
But you can also wait as the version will be released asap.

AhmedSalehz, 2025-05-08:
> > How do i get this beta version?
> 
> Beta testing is an open testing, users can assign themself in playstore. But you can also wait as the version will be released asap.

Many thanks! I will wait for it then.

reijoahola-nightscout, 2025-05-08:
> > Trend calculation was turned off.
> 
> Even if you turn it off, it will used, as long the source does not provide a trend. So this setting is ignored, if AAPS does not provide a trend.

OK, this explains that the problem looks like "fixed" :)

closed 2025-05-13 by pachi81