pachi81/GlucoDataHandler #217: AccessibilityService

IzzySoft, 2025-05-13:
Starting with release 2.0, the app declares using `android.accessibilityservice.AccessibilityService`. I could not find any explanation for that in the release notes. Could you please clarify why this is needed, and what it is used for? Thanks in advance!

pachi81, 2025-05-14:
You are right, I only have written to Playstore:

Accessibility Service API (Optional Feature):
GDH optionally uses the AccessibilityService API to display your glucose values directly on your Always On Display (AOD) screen. This feature is optional and requires a device that supports AOD. This permission is ONLY used to draw the glucose information onto the AOD. NO other data is accessed, collected, stored, or shared. The user has to grant this permission EXPLICITLY in the settings.

IzzySoft, 2025-05-14:
Thanks, included with the app's "green list" here now:

[image]

That said, maybe you want to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) and add it next to the lonely PlayStore one on your Readme, pointing it to https://apt.izzysoft.de/packages/de.michelinside.glucodatahandler ?

pachi81, 2025-05-14:
> That said, maybe you want to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) and add it next to the lonely PlayStore one on your Readme, pointing it to https://apt.izzysoft.de/packages/de.michelinside.glucodatahandler ?

Good point! I will add. I have to rework the Readme, so I can also add this.
One remark to your page, the screenshots are outdated and wrong, as I had to remove the Android Auto part. 
[Here](https://github.com/pachi81/GlucoDataHandler/tree/master/images/playstore) you will find newer ones, but not the 2 years old ones...

pachi81, 2025-05-14:
Have you thought about adding https://github.com/pachi81/GlucoDataAuto to your repo? 
I can not add it to PlayStore, so maybe it can be updated using your repo...

IzzySoft, 2025-05-14:
> Good point! I will add. I have to rework the Readme, so I can also add this.

Thanks!

> One remark to your page, the screenshots are outdated and wrong, as I had to remove the Android Auto part.

They are taken from your fastlane (that's what it's for). We cannot pull them manually for 1300+ apps on each update 😉 So once you've updated them there, they'll get pulled along with the next release.

> Have you thought about adding https://github.com/pachi81/GlucoDataAuto to your repo?

No, as there's no source code in that repo. We'd need the APK and the source code it was built from in the same repo. If the sources are in this repo here, we could pick the APK from here as well:

* if releases of both apps happen simultaneously, APKs can be told apart by their names (you already use specific naming patterns to make that possible)
* if releases are asynchronous, that could be dealt with by tag name prefixes, e.g. `gdh-1.2.3` for GlucoDataHandler, and `gda-1.2.3` for GlucoDataAuto.

If the sources are not here, please add them there. Then it could be added here (scan results look similar to GlucoDataHandler, so we can probably take the green-list along?).

pachi81, 2025-05-14:
The code is in this repo, as I use the same code basis for both apps. But the release versions and the docu is part of the other repo.
So it will not work, if the releases are in the other repo? 
Yes, it is asynchronous...

IzzySoft, 2025-05-14:
> So it will not work, if the releases are in the other repo?

Currently not, sorry – unless we forego automated updates, which is strongly discouraged. APKs are fetched from the releases in the repo defined by `SourceCode`, and sources are of course fetched from the very same repo. We currently have no separate field defined (as IzzyOnDroid started as plain binary repo about 10 years ago, there was no need for that). We might do that in the future, but there's no ETA for that (I just can say it's unlikely to happen within a year or so, our current task list is already well filled).

> Yes, it is asynchronous...

So tag name prefixes could work, if it would be OK for you shipping the APKs here (we'd need them for both, GDH &GDA, then, though). As for Fastlane, the location can be configured at IzzyOnDroid, so that would not be a problem either.

pachi81, 2025-05-20:
Hi,
just a question to the fastlane screenshots. There were 01 and 02 as **jpg** and the others as **png**.
If I change them, must it in the same format, as I changed 01 and 02 also to png!?

GDA I will not move or change, so we can let it as it is...

pachi81, 2025-05-21:
You are right, the jpg is 10 times smaller than the png. So I will also convert to jpg.
I can also convert all pictures in the fastlane!?

IzzySoft, 2025-05-21:
Of course you can. 2 possible approaches:

* you keep the "first names". Then you need to tell me once they're out which ones to clean up manually.
* you use new "first names" (e.g. prepending another `0`). In that case, cleanup would happen automatically.

pachi81, 2025-06-04:
Hi @IzzySoft,

I also change the other pictures to jpg, but the old ones are still present, even if they do not exists anymore in the fastlane.
Does it needs a manually removal or should I use png for these ones?

Thank you!

Michael

IzzySoft, 2025-06-04:
If they have the same "first name": currently yes, as mentioned above (so if you e.g. replace `01.png` with `01.jpg`, we'll end up with both – but if you replace it with `001.jpg`, the PNG would be removed, as would an existing `01.jpg` then if it's no longer in your fastlane). This is due to the fact that, for optimization, PNGs are sometimes converted to JPG at import.

I've just removed the 3 old PNG files now, will take effect with the next sync.

pachi81, 2025-06-04:
Ok, thank you for your support!!!
I will close this issue.

closed 2025-06-04 by pachi81