pachi81/GlucoDataHandler #219: Rel. 2 not working with Pixel Phone and Pixel Watch

ralf-ms, 2025-05-16:
Hi,
after upgrading to version 2 on phone (Pixel 9) and watch (Pixel watch 3), complications on watch don't get updated anymore.
"Connections" in GDH on phone and watch  showing "Awaiting data" for watch/phone.
Source is Juggluco.
I've rebooted and reinstalled GDH on both devices several times which didn't solve the connection issue.
Workaround:
reinstalled version 1.3.1 on phone (not on the watch) which immediately results in displaying complication values on the watch.

pachi81, 2025-05-16:
I haven´t heard about such problems, yet. Have you saved the logs? Have you checked the Pixel Watch app, that all permissions are granted and not battery optimization is enabled!?

ralf-ms, 2025-05-16:
Yeah, I think that permissions are not the issue.
Logs are attached, I can not make anything out of them...

[GDH_PHONE_APP_20250516_115540.txt](https://github.com/user-attachments/files/20242322/GDH_PHONE_APP_20250516_115540.txt)
[GDH_WEAR_APP_20250516_115543.txt](https://github.com/user-attachments/files/20242321/GDH_WEAR_APP_20250516_115543.txt)

pachi81, 2025-05-16:
Ok, thank you! I see a problem on the watch version, it can not decode the message. Really strange, as I only use default content, like string and integer. Seems to be a problem of the SDK. I have to check...

pachi81, 2025-05-16:
Hi, if I provide you a phone and watch version, can you install it manually for testing? But has to be installed on phone and watch....
Not sure, when I'm ready, I still working on, but it seems that I have to completely change data format for communication...

ralf-ms, 2025-05-16:
Yes, I've just checked that sideloading apk's to the watch works.
If you have anything you want me to try, let me know...

pachi81, 2025-05-17:
So, I created and tested a new version:
https://drive.google.com/drive/folders/1SFBdrZBXvgZwPBfHaiCpssyuxZOXZwRk

You have to update watch and phone, better start with watch, if you have any troubles...

ralf-ms, 2025-05-17:
Hi Pachi, 
Progress!
On watch values get updated in the application, complications not yet and the graph is missing too (intentionally?)
Overall the communication seems to be working: "awaiting data" has been replaced with the battery status on both phone and watch, nightscout is also selectable as source on watch.
Keep it up...! ;-)

ralf-ms, 2025-05-17:
Logs...

[GDH_WEAR_APP_20250517_112709.txt](https://github.com/user-attachments/files/20264039/GDH_WEAR_APP_20250517_112709.txt)

pachi81, 2025-05-17:
Have you restarted your watch? This may help with complications...

ralf-ms, 2025-05-17:
No/Yes 
I had to reconfigure the watch face complications. Obviously those settings were lost.
No it is working!
Thank a lot!
If further testing is necessary, let me know!

pachi81, 2025-05-17:
Perfect, just use it and check, if there any problems. I'm also testing in the moment.

ralf-ms, 2025-05-18:
works seamlessly, no problems so far.

pachi81, 2025-05-18:
> works seamlessly, no problems so far.

Thank you! I also have not seen any problems so far.
But I will rework it to be compatible to the old format, because a breaking change may cause unhappy people not understanding it...

closed 2025-06-02 by pachi81