pachi81/GlucoDataHandler #227: speak alarm

source2k, 2025-06-12:
I have a suggestion for a new feature for this great app.
Optional user configurable spoken TTS output for an alarm.
So it could be possible to speak something like "low 3.9".

pachi81, 2025-09-13:
> So, it's nice that you not discarded the suggestion :)

I already have put it on my todo list for version 2.2 or 2.3...

source2k, 2025-09-13:
> > So, it's nice that you not discarded the suggestion :)
> 
> I already have put it on my todo list for version 2.2 or 2.3...

That's great. Don't feel pushed, development of a good application takes time.

pachi81, 2025-10-02:
> In a sitation with the 2 version of GDH (second version) used in parallel, it would be great to have more than one TTS Voice zu choose from (M/W/D). This could help to assign the spoken values easyily to the right person. Many thanks!

This is already on my list for version 2.2 and it also effects widgets, notifications and of course TTS.
The task is like in GDA to add a name, which is shown and spoken...