pachi81/GlucoDataHandler #236: Notifications beta parsing error

chawbs, 2025-09-02:
I started getting a few random "extreme low" events over the last few days and finally worked out what it is...
GDH has an issue with the LibreLink notifications about the sensor ending soon...
Current notification is "Sensor ending soon, Replace Sensor in 37 minutes" and the previous errors would be similar but something like "replace sensor in 3 days"

I have attached a screenshot to show the handling error, and the resultant BG level was roughly BG level minus 4 mmol which did cause mild confusion at the time :)

Switched to using the nightscout connection via Gluroo app which is also installed on my device (That works well and saves me a lot of battery life on my watch

Device1: Samsung Galaxy S24 FE, One UI 7.0, Android 15
Device2: Samsung Galaxy Watch 4, (SM-R870), One UI 6.0, WearOS 5.0
GDH: 2.0.1
LibreLink: 2.12.1.11476
CGM: Libre 2+

[image]

-DJ

pachi81, 2025-09-02:
Hi,
as Libre does not provide values in notification it is not an issue of GDH, is it an issue of your configuration.
You can connect your Libre on different ways to GDH, see: https://github.com/pachi81/GlucoDataHandler/wiki/Sensors
But notification is not possible...

pachi81, 2025-09-02:
GDH also supports Nighscout as source, if you already use it...

pachi81, 2025-09-02:
So, you can try to read the notification of gluroo, but as you are also using IOB and COB from nightscout, it is not needed, as the notification reader does not support to read COB from notification (as I haven´t seen any notification providing this information, yet).

Just for interest, can you try to read the Gluroo notification with GDH and check if it is working?

chawbs, 2025-09-02:
Yes it does work. I initially had it connected that way but because it does not support COB in notification I looked at other ways.

Gluroo now host a NightScount connection so that works very well.

I use the LibreLink (horrible thing) because the data is shared with my diabetes centre (NHS UK). I am trying to convince them to use other methods too :)

Initially I had Python Scripts running on my home lab getting the data from LibreLink to produce graphs I can view from anywhere but when I saw people like Gluroo and GDH already link to watch etc, I left that behind.

GDH watch with DMM watchfaces uses so much less battery than Gluroo processes - Watch lasts all day no matter how much I use/view it without having to switch low power mode on! Awesome!

pachi81, 2025-09-02:
Thank you for testing. You can also use both, notification reader for getting current value from Gluroo (offline) and Nightscout for getting IOB/COB. But as Gluroo already needs an internet connection, it does not make sense... :-)

Juggluco also supports upload data to LibreView, so your diabetes centre can follow you. It also supports upload to Nightscout including COB and IOB, but maybe the interface of Gluroo is better than the one from Juggluco. 
(I never used Gluroo, but I have heard, that it uses a lot of watch battery)

Is there any other possibility, Gluroo shares its data? Any broadcast? But as this is software by a company, I doubt they will share their data offline...

chawbs, 2025-09-02:
Yes Gluroo can share data, but diabetes centre has to have a Gluroo setup too (set themselves up as a practice or clinic) but not sending back to LibreLink... but definitely going to look into Juggluco!

The interface for NightScout in Gluroo was very simple. Click settings, Click connect NightScout, copy and paste the URL, API Token & Secret and it all worked - but it does add in a delay because the LibreLink reads the CGM, saves to cloud, Gluroo Reads from cloud, saves to database and makes available to NightScout API...

They are experimenting with "Gluroo Local Integration" which includes broadcast integration for xDrip, BYOD and "others" - but that is still experimental so far.

So anything which will shorten the chain of data passing will be good :)

pachi81, 2025-09-02:
Great! For GDH you have to enable Glucodata Broadcast. Maybe you are also using it for Gluroo. 
But Juggluco does not share COB, only IOB (via internal webservice). If you want to use COB in GDH, you will need Nightscout at the moment.

But I´m thinking about add possibility to read COB also from notification...

pachi81, 2025-11-12:
I will close this issue.
For your information, I plan to support Gluroo COB reading in one of the next versions.

closed 2025-11-12 by pachi81