pachi81/GlucoDataHandler #242: Build for 2.1 fails

IzzySoft, 2025-09-18:
Build fails with'

```
FAILURE: Build failed with an exception.

* Where:
Build file '/build/repo/common/build.gradle' line: 19

* What went wrong:
A problem occurred evaluating project ':common'.
> Parameter specified as non-null is null: method com.android.build.gradle.internal.dsl.BaseFlavor.buildConfigField, parameter value
```

APK claims to have been built from 2bf31c9509871b77979fde46e11281c28238fcd5 – but the release tag points to a much later commit, applying a load of changes, so I don't even know at which version of the indicated file I'd have to look.

This means, the RB shield won't come up for this release. Please, always build the APK from a clean tree at the commit the release tag points to, otherwise we have problems getting RB confirmed (and it costs us a lot of extra time – with 10..20 RB builds each day, we cannot provide that time for each app). Also see: [hints for developers on reproducible builds](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Reproducible-Builds/RB-Hints-for-Developers).

pachi81, 2025-09-18:
I was also surprised, that I could not merge, as I had to rebase first. But as it was only a change to readme in master, I did not care....

closed 2025-09-18 by IzzySoft

IzzySoft, 2025-12-10:
Err… Please do not build from non-existing commits (or, do not let the commit you built from disappear)! v2.2 was built from 73326aba3bef17074db664b3d264919181b741ba – but that commit does not exist here. Which is not the only problem: even if I'd compensate for that, there are still differences in `classes3.dex` when building from the tag, which then cause differences in the baseline. Just checksums, strangely; did you change something in the build process? Did any of the `local.properties` change and might have caused this?

Currently, v2.2 must count as "RB failed". I will keep those results back for a day (until tomorrow's run) in case you have some idea what could be done, @pachi81

pachi81, 2025-12-12:
What can I do to prevent such problems? Using always the same version to build the apks? Even if I´m using a newer version for developing?
I already thought about some script to create the final version, but had no time, yet.

What can be the affect of such a class-diff?

pachi81, 2025-12-17:
That is strange...
I'm using the version checker only for GlucoDataAuto. I need to check.
I added a webserver and the health connect feature, maybe the health connect can cause such problems!?

pachi81, 2025-12-17:
This is the password checker. Is part of the code, but not used:
https://github.com/pachi81/GlucoDataHandler/blob/master/common/src/main/java/de/michelinside/glucodatahandler/common/utils/GitHubVersionChecker.kt

I can also move it to Android Auto part, so it is anymore in the common section. 
Then I can provide you a version without for checking!?

IzzySoft, 2025-12-18:
Limiting to 8 cores did not do the trick. OK, let me check the build log for hints:

```
> Task :mobile:compileReleaseKotlin
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/MainActivity.kt:971:35 'fun getColor(p0: Int): Int' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/notification/LockscreenActivity.kt:278:20 'fun setDecorFitsSystemWindows(p0: Boolean): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/AppSettingsActivity.kt:101:17 'fun setTargetFragment(p0: Fragment?, p1: Int): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/ExportImportSettingsFragment.kt:54:17 'fun startActivityForResult(p0: Intent!, p1: Int): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/ExportImportSettingsFragment.kt:64:17 'fun startActivityForResult(p0: Intent!, p1: Int): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/ExportImportSettingsFragment.kt:84:18 This declaration overrides a deprecated member but is not marked as deprecated itself. Add the '@Deprecated' annotation or suppress the diagnostic.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/ExportImportSettingsFragment.kt:88:19 'fun onActivityResult(p0: Int, p1: Int, p2: Intent?): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/LogSettingsFragment.kt:119:13 'fun startActivityForResult(p0: Intent!, p1: Int): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/LogSettingsFragment.kt:126:18 This declaration overrides a deprecated member but is not marked as deprecated itself. Add the '@Deprecated' annotation or suppress the diagnostic.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/LogSettingsFragment.kt:130:19 'fun onActivityResult(p0: Int, p1: Int, p2: Intent?): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/SettingsFragmentBase.kt:307:37 Condition is always 'true'.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/preferences/SettingsFragmentCompatBase.kt:33:28 'fun setTargetFragment(p0: Fragment?, p1: Int): Unit' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/watch/WatchDrip.kt:86:28 'fun <T : Parcelable!> getParcelable(p0: String?): T?' is deprecated. Deprecated in Java.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/widget/FloatingWidget.kt:89:14 'open' has no effect on a final class.
w: file:///build/repo/mobile/src/main/java/de/michelinside/glucodatahandler/widget/LockScreenWallpaper.kt:82:14 'open' has no effect on a final class.

> Task :mobile:compileReleaseJavaWithJavac
Java compiler version 21 has deprecated support for compiling with source/target version 8.
Try one of the following options:
    1. [Recommended] Use Java toolchain with a lower language version
    2. Set a higher source/target version
    3. Use a lower version of the JDK running the build (if you're not using Java toolchain)
```

Are you using JDK 21, or 17? Checking our recipes, we've established GlucoDataHandler with JDK-21 right away.

```
> Task :mobile:checkReleaseDuplicateClasses
WARNING: [Processor] Library '/build/.gradle/caches/modules-2/files-2.1/androidx.car.app/app/1.7.0/368eb3df2522b17368912c34129ebf3c1f0f9519/app-1.7.0.aar' contains references to both AndroidX and old support library. This seems like the library is partially migrated. Jetifier will try to rewrite the library anyway.
 Example of androidX reference: 'androidx/car/app/media/IMediaPlaybackHost'
 Example of support library reference: 'android/support/v4/media/session/MediaSessionCompat$Token'
```

Do you see the same warning in your logs?

OK, tried with JDK-17. Similar outcome, comparable log entries. Just the one for `Task :mobile:compileReleaseJavaWithJavac` looks different now:

```
> Task :common:compileReleaseJavaWithJavac
Note: /build/repo/common/src/main/java/com/eveningoutpost/dexdrip/services/broadcastservice/models/GraphLine.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /build/repo/common/src/main/java/com/eveningoutpost/dexdrip/services/broadcastservice/models/GraphLine.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
```

Maybe let's compare the environments. We just call `./gradlew :mobile:assembleRelease` after having prepared the `local.properties`. Base image used was `ubuntu:jammy` (we can switch to `ubuntu:noble`, or Debian bullseye/bookworm/trixie if that would help), OpenJDK 21 (17 just for this test run, though we can set that if needed). Build happens in `/build/repo`, which is the root of the checked-out repository (we can adjust that, too). What is your setup?